mtongnz/ESP8266_ArtNetNode_v2 #15: WS2812 not working properly

mtongnz, 2017-02-08:
I've found that the ws2812 doesn't work properly when using different power supplies for the ESP and the pixel strip.

Try running the device with the same 5V supply you use for your strips.  Obviously don't put 5V direct into the ESP - use a 3.3V reg to drop it down.

This seems to only be an issue with a larger number of pixels.

Please leave some feedback if you experience this

mtongnz, 2017-02-11:
Currently testing some new timing at a slower rate.  If it works, I'll add an option to the web UI so you can have the fast aggressive timing or the slow stable timing.

It seems to only be an issue with longer strings of pixels

nanu-c, 2017-02-12:
I think the solution for that is in experience to connect the - pol of the two power supplies.
The ws2812 is really sensible in that case.
And other thing i experienced with long strips is the voltage drop on the signal channel. You can use pull-up resistors to solve that or don't use more than 1m of cable in between the end of the last stripe and the first of the next stripe...

mtongnz, 2017-02-12:
beta4a has slightly slower data rates to try to help with this.  Initial tests seem to show good results

kaeltaz, 2017-02-14:
Try this timing
[image]

mtongnz, 2017-02-19:
Beta5 has much more strict timings in use.

It's working flawlessly on my 150 pixels - a few more are on the slow boat from China thanks to a couple of generous donations I received.

Try it and let me know.

mtongnz, 2017-02-26:
#36 is the same issue as this.  I am closing this thread and we can discuss there

closed 2017-02-26 by mtongnz