mtongnz/ESP8266_ArtNetNode_v2 #16: Feature Request: Scene recall with buttons

mknerr, 2017-02-08:
Was wondering if it would be possible to add buttons that would trigger scenes.

mtongnz, 2017-02-09:
This is definitely a possibility.  There will be a button added in the future to reset the factory default settings (incase you forget a WiFi password or set an invalid IP).
I should be able to utilize the same button to select scenes.  I still have to implement scenes in this firmware yet so it might be a while.

mknerr, 2017-02-09:
Thanks for the reply and all your work on this. Have ordered parts so we'll see how it goes. Will just be for a small band lighting system driven by Luminair. Was thinking the presets might be nice for a backup or for perhaps situations that are basically static.