mtongnz/ESP8266_ArtNetNode_v2 #17: First pix on 2 uni is blinking

kaeltaz, 2017-02-09:
First pix on 2 uni is blinking

[https://youtu.be/qbqkvZ2hWWU](url)

mtongnz, 2017-02-11:
I believe this is related to this issue: #15 

I am waiting on some new pixels to arrive (bought thanks to a couple of donations I've received) before I can fully test my fix.  At present I only have 150 pixels to test with :(

mtongnz, 2017-02-12:
I've released beta4a.  It has a slightly slower data rate which will hopefully fix this issue.  Please try and let me know

Alfiszcze, 2017-02-17:
I belive that it is universe problem because the same issue is with 1st pix on 3rd universe .
512 dmx /3 = 170 and leftover 2 slots from dmx universe.

mtongnz, 2017-02-19:
Please try beta5 and see if it's better

mtongnz, 2017-02-21:
OK.... I've actually found the cause of this.  It was nothing to do with timing - I was accidentally writing over the first 2 channels of uni 2 with channels 511 & 512 of the first uni.

beta5a should definitely fix this issue.  I've only got 150 pixels so please test and let me know.

closed 2017-02-26 by mtongnz