mtongnz/ESP8266_ArtNetNode_v2 #22: Web Ui

kaeltaz, 2017-02-13:
My suggestion on web ui

[image]

mtongnz, 2017-02-13:
Ability to add and select custom themes will be added once I implement SPIFFS for storing scenes

kaeltaz, 2017-02-13:
I thought about the main themes

mtongnz, 2017-02-27:
You can now upload your own themes to the device in beta5c.  Goto /style to upload a new theme or revert to the default.

If anyone comes up with some good themes, please send them to me.  I'd love to have a look

Wachenanggo, 2023-01-02:
> My suggestion on web ui
> 
> [image]

Where can i customize that web ui?? or app?