mtongnz/ESP8266_ArtNetNode_v2 #23: WS2812 RGB is swapped to GRB

querex007, 2017-02-15:
WS2812B RGB is swapped to green red blue.
Most artnet software use the RGB sequence as default.

Guess this is only a minor fix.
Keep up the awesome work 🙈💪

mtongnz, 2017-02-16:
I'm aware of this.  I just change it in the sending software right now.  

It's not a bug as such, more just that the pixels use GRB instead of RGB in their data sequence - I don't know why.

I plan to add support for different configs at some point shortly.

mtongnz, 2017-02-16:
That was my thoughts - it's extra processing that isn't really needed.  I may add it if it doesn't cost too much in overheads

mtongnz, 2017-02-20:
After some consideration and talking with a few others' I've decided that I won't be implementing this in the foreseeable future.

querex007, 2017-02-20:
I get the point...
But one more comment, just to discus the project to make it as best as possible :-)

In point of use, RGB is more common use in the light and sound business, so also more user friendly instead of GRB.
(Everything is RGB of CMY in the light business).
The majority of the users of this project will control this WS2812 in RGB instead of GRB.

I guess it's the best to implement everything as RGB.
And I confirm, add an additional config property on the web GUI is overkill.
For the minority of users who want to use the controller is combination with GRB over artnet, they can chance the config of there artnet control software.

kaeltaz, 2017-02-21:
WS2812 is GRB as invented manufacturer. Most control software (Chamsys, MA2, Vista, Avolites, freestyler, JinX, Madrix can be replaced GRB to RGB control. This is no problem.

querex007, 2017-02-21:
@kaeltaz / @mtongnz 
It's strange 'the use of a product' (DMX / artnet channels) needs to be aligned to the internal datastream (arduino > WS2812)?
As the general use of the product is RGB in 99,99% of the time.

**This sounds like:**
When the (internal) communication from your car it's steering wheel is swapped (left = right, right = left) over the CAN-bus.
They should make the steering wheel also swapped for the end user (left = right, right = left)?
You got the point?

Update the head fixture in your light control software is no big deal (RGB > GRB), but no user friendly...
When you have a lot of WS2812 strips, it will be annoying. :smile: 
The LED standard is RGB in DMX and artnet.

Erates, 2017-02-21:
+1, to follow the RGB standard as artnet/dmx color sequence

mtongnz, 2017-02-21:
As has been said above by a few others, any decent control software will have GRB ordering built in.  Instead of adding an RGB fixture, select the GRB.

I don't get why they made ws2812 GRB instead of RGB but we can't change it now.  I get that RGB is the standard (I've been a lighting tech for 20 years) but performance is more important than a feature that can be handled by the controller.

**Adding this setting will use valuable clock cycles during the incredibly time critical ws2812 output.**

Most libraries that implement this feature do it as assignment stage - ie. you set an LED colour and it swaps them around and stores it to the buffer.  As all the data is coming in one block from Artnet for this project, it would require an extra step of reordering that block.  It would be a time/resource consuming process.  As it can be done by the (far more powerful) lighting controller, I don't think it is worth it.

querex007, 2017-02-21:
@mtongnz 
Thanks, for the more in depth description.
I thought there was already a configured mapping from artnet > ws2812 😄
I understand the technical issue with the valuable clock cycles, you are correct!

I'm an ex lighting tech (9 year active), but I'm stil using Chamsys for some smaller private projects / events :roll_eyes: I can't live anymore without DMX / artnet :rofl: (master-slave or a Christmas tree of random scrolling colors, oh boy!).

Chamsys doesn't have an "GRB" head in it's (generic) default library:
LED 7chan, LED AWUV, LED RGBAW, LED RGBA, LED RGBW, Led10bar, Led20bar, Led3bar, LED3ch, Led4bar, Led4by2, LED4ch, Led5bar, LEDDRGBS, LEDRGBCD, LEDRGBDS, LEDRGBDSM, LEDRGBMS, LEDRGBSD, LEDRGBWDS, LEDRing, LEDWhite

Duplicate the RGB head and swap the color sequence to is not an issue.
But my advice was to improve the project, to support general users (with less lighting desk knowledge).

mtongnz, 2017-02-21:
I haven't used Chamsys much but a quick search turned up this:
https://www.controlbooth.com/threads/magicq-rgb-grb.34803/
You should only have to change it once to make a new fixture, then just patch as many as you need.

I do appreciate the feedback and I had seriously looked into doing this.

The best way I could think to do it was to read the bytes in a different order when outputting - this would save a long for loop when receiving Artnet.  Unfortunately this would mean less time for interrupts to run during ws2812 output - this is needed for DMX in/out to function.  At present, I can only allow a 3ms interrupt in my DMX library or the ws2812 timings get screwed up.

When I release the source in a month or so, maybe someone can come up with a good solution.

kaeltaz, 2017-02-21:
@querex007  for sure?
[image]

kaeltaz, 2017-02-21:
I use chamsys for 5 years. I also tests the this Artnet and ws2812 :)

closed 2017-02-26 by mtongnz