mtongnz/ESP8266_ArtNetNode_v2 #26: Hotspot SSID

kaeltaz, 2017-02-19:
Hotspot SSID do not contain last digits of the MAC address

mtongnz, 2017-02-19:
???

kaeltaz, 2017-02-19:
This is not last digits of the MAC address

[image]

mtongnz, 2017-02-19:
It's not meant to be.  It's a 5 digit number derived from the chipID (which the ESP8266 does use to generate it's MAC address).

The purpose of the number is to ensure that if there are multiple units, they will have a unique SSID.  It will stay the same for the entire life of the unit as the chipID never changes.

This method is only used when the default hotspot SSID is in use.  If you change it, the number wont be used.

mtongnz, 2017-02-19:
This is the code used to generate the SSID:

`// If it's the default WiFi SSID, make it unique
  if (strcmp(hotspotSSID, "espArtNetNode") == 0 || hotspotSSID[0] == '\0')
    sprintf(hotspotSSID, "espArtNetNode_%05u", (ESP.getChipId() & 0xFF));`

closed 2017-02-20 by mtongnz