mtongnz/ESP8266_ArtNetNode_v2 #27: Can'tupload firmware again.

asm7100, 2017-02-19:
Hi
Grate job :-)

I install the firmware with the http://esp8266-webupdate.local works fine the first time.
Made a error in the setup, and can't connect to it again, so i try to do the reset (power of min 5 times) and The GPIO14 to GND at start up, nothing works.
Then i try http://esp8266-webupdate.local to upload the firmware again.
Now i only get a lot of "noice" on the com port, and no SSID to connect to??

Try it a loot of times, nothing works. If i upload anoter project it all works fine, but not with your firmware.

I'am using Wemos D1 Mini.

asm7100, 2017-02-19:
The firmware will not run at all, its like the setting is fuct up

asm7100, 2017-02-19:
Yes i will trye the new Wemos D2 bin now.

mtongnz, 2017-02-19:
Is it working?  Or at least running?

asm7100, 2017-02-19:
Thats works, but when i give it a ip address in the setup, like 192.168.3.100 and there is no GW address to set. I can't connect to it any more.

mtongnz, 2017-02-19:
Gateway isn't needed as we're not connecting to the internet.  Make sure you've set the subnet correctly.

Also, if you're trying to change the IP for hotspot mode, it won't work.  The IP in hotspot mode is always 2.0.0.1.

I haven't tested the static IP settings for a while.  I'll check them later to see they are working.

asm7100, 2017-02-19:
I have my local network with wifi, an i normel use 192.168.x.x
And my wifi router is at 192.168.0.1 where all my computers are on.
i like the ArtNetNote to be on 192.168.3.100 and forward.
Can't i connect to the setup / config page by using 192.168.3.100 ?

mtongnz, 2017-02-19:
Yeah you can but you'll need to setup some routing tables or static routes in your router. I'm not very good at this.

Easiest way would be to set your computer IP into the 192.168.3.x subnet.

asm7100, 2017-02-20:
Why not just add a gw i have 6 security cam, 3 computer, som tables, oure phones, NAS server, some other ESP devices printer and so on.

asm7100, 2017-02-20:
When will you release the code, so i can make it my self?

mtongnz, 2017-02-20:
Code release will be when I do the first non beta release.  I still have sACN support to add first.  I will look into adding gateway for the next beta also.
This will probably be a few weeks away yet.
I am closing this issue as the initial issue is closed - I will open a new one with the gateway feature request

closed 2017-02-20 by mtongnz