mtongnz/ESP8266_ArtNetNode_v2 #29: GPIO2 on Wemos D1 mini is not good.

asm7100, 2017-02-20:
GPIO2 on Wemos D1 mini is not good, it has a 10K pull up,
Maby use GPIO5, GPIO4, GPI016, GPIO12, GPIO13 or GPIO14 
Or better let the user set op up for ports.

asm7100, 2017-02-20:
In you sch you are not using GPIO4 maby use that to Port B

mtongnz, 2017-02-20:
That's true but I use it for my scope for testing.  I'll add it as a compile option shortly

asm7100, 2017-02-20:
Then i will test it :-)

mtongnz, 2017-02-20:
Is this for TX_B?  I can't change that - it's in the ESP8266 hardware which pin it's on.  I thought it was for a direction pin...

Note that the Wemos pins are labelled similar to the NodeMCU pins - it's the wrong numbering from the ESP8266's point of view.  You should be using this:

TX (GPIO1) for DMX_TX_A
D4 (GPIO2) for DMX_TX_B
RX (GPIO3) for DMX_RX (both A & B)
D1 (GPIO5) for DMX_DIR_A
D0 (GPIO16) for DMX_DIR_B

mtongnz, 2017-02-20:
I've added this info to the readme

asm7100, 2017-02-20:
D4 (GPIO2) Has a 10K pullUp and a led on it, Not good, dosent work.

mtongnz, 2017-02-20:
This is a hardware limitation, not software. Unfortunately they've designed their hardware poorly. Changing this in software will break other features of my firmware. Maybe just remove the LED and resistor.

asm7100, 2017-02-20:
Way not just swap to pin's
In the readme filen 
WEMOS: This is compiled for Wemos D1 and similar boards. It uses the 4M (3M SPIFFS) compiler option.
How about specifi you can only use Part A to (DMX) nothing more.

asm7100, 2017-02-20:
Way will it break other features in your firmware?

mtongnz, 2017-02-21:
I won't be adding this feature. This device is primarily fir dc and rdm. If you want both outputs to be ws2812 then remove the LED and resistor.

I plan to make a similar unit to this just for ws2812 output in the future with 4-8 outputs.

closed 2017-02-21 by mtongnz

MikeCurrington, 2017-06-08:
As a follow-up to this bug, the D4 (gpio2) port on the Wemos D1 mini works fine as an output for DMX data.
Asm7100 is correct that there is an led on the same port but unless there is another library trying to control the led this is not an issue.
I tested using an official Wemos D1 mini.