mtongnz/ESP8266_ArtNetNode_v2 #32: OLED and Encoder

kaeltaz, 2017-02-21:
My proposition to use oled and encoder instead of button,led. Oled give more opportunities, info and change the settings. 
[image]
[image]

kaeltaz, 2017-02-21:
oled is cheap, only costs $2.80 (AliExpress)

mtongnz, 2017-02-22:
I like this idea. Out would have to be interrupt driven.

I haven't worked with encoder wheels before.  Would the 2 pins provide a click button also?

kaeltaz, 2017-02-22:
2 pins for encoder
1 pin for click button

[image]

mtongnz, 2017-02-25:
I am curious how it would go having the oled running off gpio0 which is used a a flash enable pin

kaeltaz, 2017-02-26:
https://nathan.chantrell.net/20141230/wifi-mqtt-display-with-the-esp8266/

gravicapa19, 2017-10-24:
I used this library:
https://github.com/klarsys/esp8266-OLED
Thanks klarsys
[image]
[image]