mtongnz/ESP8266_ArtNetNode_v2 #34: Testet on Wemos D1 Mini

asm7100, 2017-02-22:
Hi now i have testet the last to days.
I dosent seems that your software / connection can work on Wemos D1 mini. :-( To bad.
Using TX pin for WS2812 just look it up and it will not even start up :-( 
And the other port, has a pull up so that dosent work eather.
I havent tryed it with DMX and that i will not do.

Now i will have a go with ESP01 and a ESP12 :-)

asm7100, 2017-02-22:
Now i got it to work, jave gave it a shorter Short Name:
and a shorter Longer name?????

asm7100, 2017-02-22:
WS2812 no port A (TX)

mtongnz, 2017-02-23:
Could you please let me know what name you were using for long and short?

My code should limit the lengths to 20 & 60 respectively as defined by the Artnet v4 standard.

Also please check the new beta5b as it has slightly different compiler options used.

asm7100, 2017-02-23:
eANNW 3-100
eANNW 3-100 - Matrix x 2

This is the two names i am using
And the new v2.0.0 (beta 5b) works fine.

mtongnz, 2017-02-26:
Does the new beta work with the longer names you tried?

closed 2017-02-28 by mtongnz