mtongnz/ESP8266_ArtNetNode_v2 #36: ws2812 flicker at random times

mtongnz, 2017-02-23:
This issue is caused by a WiFi/system interrupt causing a low of >5us to be sent to the ws2812 data line.  I am still trying to find a way to stop this interrupt.

In the mean time, I have made a work around by disabling all interrupts during ws2812 transmission.  This however doesn't work when using DMX as the DMX needs interrupts to function.  

If you are only using ws2812 strings, ensure both ports are set to ws2812.  If you are using DMX & ws2812, you may experience flickers for runs of longer than about 150-200 pixels.

TheSeven, 2017-05-02:
Have you considered using the I2S hardware to drive WS2812 in the background instead of bitbanging it?
Have a look at https://github.com/TheSeven/esp8266firmware/tree/master/src/device/ws2811 for an example.
Some of that code was written by me (feel free to use that), but the I2S interface setup etc. was cherrypicked from random places all over the internet - not sure where exactly the idea originates from. It's been floating around for a while.
Once I get around to getting this project set up on physical hardware (sorry, I'm quite busy lately as well), I might have a go at implanting that into this project.

scamiv, 2018-03-20:
I needed multiple stripes per esp so I2S wasn't an option.

The reason for the flickering is an added 6µs by the wifi IRQ. However this wont be disabled by calling RSIL as it is managed by the NMI.
Therefore you also have to mask the non-maskable interrupt to get consistent timing.
Luckily some wise guys found out how: https://www.esp8266.com/viewtopic.php?f=9&t=3979&start=20
After some trial and error i got it to work, it a'int pretty yet but flicker free at last.
[https://gist.github.com/scamiv/b931b307afaead5a0f43e6eb2bc57e08](url)