mtongnz/ESP8266_ArtNetNode_v2 #41: WDT Resets

mtongnz, 2017-02-28:
There seems to be an issue with the watch dog timer causing resets about 10 minutes into operation.  I haven't been able to track it down and it takes ages to fault find when I've gotta wait for it to happen.

If you are watching in DMX Workshop, you'll see this show up in the status field when the device resets.  The output will also freeze for about 10 seconds.

Anyone experiencing these, please post your settings and the counter from status in DMX Workshop so I can see if there is a pattern.

theudas, 2017-09-15:
Does it happen always or only using leds?

ammuller, 2017-09-26:
Could you please try my code, I don't get WDT resets.
The lights stop working after a while, a reset from the web interface fixes it.
https://github.com/ammuller/ESP8266_ArtNetNode_v2

ElToberino, 2017-12-03:
I'm using your really great software with Ws2812b Pixels on Port A and Port B. (about 900 pixels in total)

After a certain time (about 20-30 Minutes) my Pixels freeze. Wifi connection is still there, I can load the web interface. I found out, that the only way to get back to normal mode is powercycling to times **or** resetting via web interface two times. It's interesting that powercycling or resetting only one time doesn't help.

I also noticed that the period of time until freeze is in hotspot mode much shorter than in Wifi-Mode.

Hope, this experiences will help solving the problem. I like your software very much and would appreciate any solition of this problem.

rsototec, 2017-12-04:
> Enviado el: domingo, 03 de diciembre de 2017 01:01 p.m.

Hi, I have the same issue. I am using the Adafruit HUZZAH ESP8266 Breakout.
Thanks for your attention.

dddaniele, 2018-03-05:
Hi bro, this solutions fix the problem for me (firmware 1.2) test 4 times for 4hour every time:
In ESP8266_ArtNetNode_DMX.ino:
- Add  ESP.wdtFeed() at void loop() start 
- Add a delay(15) after webServer.handleClient()
I found also another bug in library that inibite the dmx "strobo funcion"
Please tell me if this work for you

bombcheck, 2018-03-05:
Damn, overlook the most obvious thing. :)
Will give this a try tomorrow.

dddaniele, 2018-03-06:
I already tried only with ESP.wdtFeed() only but wdt reset comes back. I also tried a delay 10 and the problem persists.
Setting instead ESP.wdtFeed() and delay(15) i have not encountered problems with my par LEDs and moving heads.

Try to comment those lines on espDMX.cpp (in the library folder) (this cause stobo function and wdt reset issues)

// Output full universe (512 channels) once every x seconds
    //if (millis() > dmx->full_uni_time + DMX_FULL_UNI_TIMING) {
    //  dmx->full_uni_time = millis();
    //  dmx->state = DMX_FULL_UNI;

    //  // Status LED
    //  if (dmx->ledPin != DMX_NO_LED) {
    //    if (dmx->led_timer++ >= 2) {
    //      analogWrite(dmx->ledPin, dmx->ledIntensity);
    //      dmx->led_timer = 0;
    //    }else
    //      analogWrite(dmx->ledPin, 0);
    //  }
    //} else

bombcheck, 2018-03-06:
espDMX.cpp? Did not find this file in the libraries. And can not find the mentioned lines anywhere in the project.

bombcheck, 2018-03-06:
Adding a delay of 15 ruins timing at the end of my strip. Reducing the value and trying some other modifications. But the issue seems to persist. It just lasts longer then before prior to crash...

dddaniele, 2018-03-06:
i'm running SP8266_ArtNetNode_DMX firmware 1.2

bombcheck, 2018-03-06:
Ahhhh I see. But here we are discussing the "new" v2 firmware which has these issues too. :)

delay (5) seems to work for me: No updating issues. Keeps reseting sometimes, but I catch the reset reason and doing a ESP.restart if the watchdog or an exeption occured.

bombcheck, 2018-03-07:
It keeps crashing from time to time. Sometimes after minutes, sometimes after hours. I THINK it may be from some bad packets received or something and not from the WDT...

ElToberino, 2018-03-28:
Does the WDT/freeze issue also occur with the old version ArtNetNode v1.2.0?
I just want to set up a simple node, without pixel support, RDM etc. So v1.2.0 would be perfect for me, if there wasn't this problem.

scamiv, 2018-04-20:
Have you given my fork a try? It should fix some/most of the existing problems.
The current git version runs stable for a few hours in most cases, however there's still the issue of heap fragmentation.
I will make another push soon which switches the lib over to umm_malloc to counter this.

idtres, 2018-06-17:
@scamiv I will check out your fork.
Can you summarize what are the main changes, and for how long did you run the node without any reset/frozen outputs?

blackrockcitynv, 2018-07-09:
scamiv.. does this fix the LED freeze and rebooting issues?

codehackr, 2018-07-15:
i tried scamiv fork of this code in my case it only last 15 mins and hard freezes requiring a power cycle to get it up and running again.

blackrockcitynv, 2018-07-17:
I have done extensive research on this issue and tested code many many different ways and setups. Its 100% the wifi services getting slammed with data.. the wifi lifetime timer does not renew when data is being sent. If i stop data send the timer renews

blackrockcitynv, 2018-07-17:
I looks like anyone and everyone who has attempted to make an Artnet wifi node has ran into the same issue (i have tested nearly every artnet neopixel wifi code on github) all the same exact wifi lifetime timeout issue

parechiotto, 2018-11-27:
Hi all!
I don't know if the bug was resolved from someone.
I've try to add thi solution: https://community.blynk.cc/t/solved-esp8266-nodemc...
And for me it seems work!

So I add in setup()
ESP.wdtDisable(); 
ESP.wdtEnable(WDTO_8S);

and in and in loop()

ESP.wdtFeed();

krupaw2, 2018-12-10:
How much I got tired of this program, I'm a beginner and I thought I was doing something wrong.
Meanwhile, after installing the "Esppixelstick" all problems have disappeared - I highly recommend.

Furbean, 2020-06-14:
Hi Matthew,

**The issue is not a software glitch**, i've discovered your project today.
I use a esp8266 (WeMos D1 R1) with 300 leds W2812s, i have a 20 ampere power supply for the 5 volts. After 10 mn, the esp8266 seems to freeze and between 3 and 10 mn, lights start again the animation.

Since i've connected the esp8266 on my powered usb port, i don't have any problems. It works very well even with a lot of effect in Jinx!. 

We must add a capacitor to protect cpu i think, i will check this week.

Thanks for you great work !!

Igor1976-light, 2020-06-14:
Hi Furbean,

do this, paste into setup ()

ESP.wdtDisable();
ESP.wdtEnable(WDTO_8S);

and in loop()

ESP.wdtFeed ();

and the problem should be resolved. At least it works for me.

And a 1000µF / 16V capacitor comes between ground and + in the power supply of the LEDs

diegoferrero446, 2020-06-22:
Sorry, none of the solutions provided is a definitive solution, the problem is in Store