mtongnz/ESP8266_ArtNetNode_v2 #42: Sk6812 led pixels

Bensmall, 2017-03-02:
Hi is it possible to implement the use of sk6812 pixels (same as ws2812b but rgbw Insted of rgb) as you get exactly 128 pixels per 512 dmx channels as configured for ws2812b only uses 510 dmx channels for 170 pixel
Many thanks Ben

kaeltaz, 2017-03-02:
[image]

mtongnz, 2017-03-04:
Thanks @kaeltaz.  Donations are much appreciated and allow me to purchase more bits for testing.  I have actually received a couple of donations in the last few days which I've used to purchase a strip of RGBW SK2812 LEDs.

I'll try to get this implemented for the next beta release so it can get tested by everyone before the final release.  The hardest bit is going to be updating the Pixel FX engine - it was designed for RGB, not RGBW so may be a bit of a challenge but we'll see how it goes.

mtongnz, 2017-03-27:
Implementing this is not quite as simple as I'd hoped - mainly due to the Pixel FX code being a bit sloppy.  I'll be releasing the source soon with a couple of rewards up for grabs for people who fix a few of the issues.  This will be one of the issues with a reward - it'll be one of the prototype boards I'm working on.

Bensmall, 2017-04-08:
hi just having a think when using the code for controlling 8 universes of
pixels does that allow me to use the full 8 universes worth of channels or
is it just the 12 channels for pixel fx ? and could it be that with the
sk6812 it uses just 8 universes with of channels if the fx code being a bit
of a pain with the sk6812 ? many thanks Ben

Deadeye5589, 2017-10-01:
What are the current problems with the implementation of this feature? Still trouble with the fx code or is it the driver itself?

mtongnz, 2017-10-10:
The driver will just need the timings tweaked for it to work.  The FX code needs to be re-written a fair bit for it to work.  I haven't had time to play with the SK6812 strips yet so haven't seen how much will need to be done.  Feel free to have a play with ws2812Driver.cpp and see if you can get the timings working.

Bensmall, 2017-10-10:
Okay will give it a go just need some time off work to have a play lol

Deadeye5589, 2017-10-11:
Timing should not be that much of an issue, since they are very similar between WS and SK LEDs. For both WS and SK the timing deviation is +/- 0.15µs. If your program has a tighter tolerance there is a good chance to have the SK LED running without modifications to the timing. 

[image]

Bensmall, 2017-10-11:
The only issue I had when using sk6812 on ws2812 timing is that when using
more than one universe of them the seconds one is 2 channels behind as 170
ws2812 pixels use 510 dmx channels and then the last 2 dmx channels are
skipped, but with sk6812 128 pixels use 512 dmx channels so when using more
than 128 pixels there is issues, all that needs doing really is to tell the
code to use all 512 channels per universe of artnet and output it as spi
data for the pixels :)

Deadeye5589, 2018-05-17:
Finally I found some time to look deeper into the code, figuring out that I need some feedback on how the code works to fully understand it. 

A new drop down selection for the web interface was created quiet easily. Similar the SK6812 value in the p_type enumeration. As far as I can see this new value should be processed by JSON quiet fine without further modifications. My blocking point is the actual call of the ws2812 drivers. I cant figure out how the output is being generated. I found functions for setting pixels, but I can't find the spot where they are called inside the program. Else I would simply double the ws2812 class and modify it for the sk6812. 
With my SK6812 strips I can't get any output from the current software. Therefore some timing modification seems necessary.

Deadeye5589, 2018-05-18:
Hmm Arduino is blocking any progress on this.
For some reasons the commented lines regarding the SK6812 will prevent the compilation of the project

void doStatusLedOutput() {
  uint8_t a[9] = {0, 0, 0, 0, 0, 0, 0, 0, 0};
  
  if (!statusLedsOff) {
    if (statusLedsDim) {
      for (uint8_t x = 0; x < 9; x++)
        a[x] = statusLedData[x] & STATUS_DIM;
    } else {
      for (uint8_t x = 0; x < 9; x++)
        a[x] = statusLedData[x];
    }
  }

  #ifdef STATUS_LED_MODE_APA106
    pixDriver.doAPA106(&a[0], STATUS_LED_PIN, 9);
  #endif

  #ifdef STATUS_LED_MODE_WS2812
    pixDriver.doPixel(&a[0], STATUS_LED_PIN, 9);
  #endif

// No more space available after this ????
//  #ifdef STATUS_LED_MODE_SK6812
//    skpixDriver.doPixel(&a[0], STATUS_LED_PIN, 9);
//  #endif

  // Tint LEDs red slightly - they'll be changed back before being displayed if no errors
  for (uint8_t x = 1; x < 9; x += 3)
    statusLedData[x] = 125;
}

void setStatusLed(uint8_t num, uint32_t col) {
  memcpy(&statusLedData[num*3], &col, 3);
}

The compiler error stated that the text' will not fit in region `iram1_0_seg but I don't see why this small change should use up all of the remaining resources. The original code only used 31% percent of the available program memory and 57% of dynamic memory.

Deadeye5589, 2018-05-25:
Good news,

both SK6812 and WS2812 LEDs are working with the same timings. So the only thing that must be realized is the channel mapping 3 for ws and 4 for sk. 
Still I can't figure out why the chip should ran out of memory by adding a sk6812 class.