mtongnz/ESP8266_ArtNetNode_v2 #43: Possible to get more than 2 LED outputs?

fjanicki, 2017-03-08:
Would it be possible to get more than 2 LED outputs. Is there any hardware restriction behind this limitation?

mtongnz, 2017-03-09:
Are you meaning status LEDs?  If so, I'm currently working on using ws2812 LEDs for status instead.  I'm just waiting on them & the prototype PCBs to arrive before I can test it.

If you're meaning Artnet controlled LED outputs, then you can actually run 1360 LEDs - 680 per output.  These are the WS2812 addressable LEDs, not a standard LED strip.

mtongnz, 2017-03-09:
Also, I plan to start on a device in a few months - once I've got this project working "finished" and released.
It will hopefully have 8x WS2812 outputs, each capable of running at least 680 LEDs.

closed 2017-03-27 by mtongnz