mtongnz/ESP8266_ArtNetNode_v2 #45: Only first LED lights up

potentstudios, 2017-03-13:
The first led only lights up green, yellow or white. I've used the same wiring (with the exception of the data pin) as this project https://github.com/scottlawsonbc/audio-reactive-led-strip and the entire LED strip lights up fine. Here are links to my patch and output settings. Patch: http://i.imgur.com/Y52UMQs.png Output: http://i.imgur.com/C2sdALC.png. For some reason Jinx! doesn't automatically recognize the nodeMCU as an output device but Vista 2 does.

jmb-dmx, 2017-03-13:
You have load the wrong project for pixel ! This is only DMX output.
This is the good https://github.com/mtongnz/ESP8266_ArtNetNode_v2

potentstudios, 2017-03-13:
Ope! When using the "Web UI OTA from my previous project" method to update the firmware with the .bin file, the nodeMCU shows a solid red light after clicking update in the Web UI.

jmb-dmx, 2017-03-14:
I have same thing with Adafruit Feather..... No problem with wemos and esp07.

mtongnz, 2017-03-14:
So this is a web UI update issue rather than pixel output issue?

It is possible that they have a different flash chip size which has caused me issues in the past.  Check your flash size - how to here: https://www.instructables.com/id/ESP8266-Artnet-to-DMX/#step4

Hopefully your chips are a 1M flash size or higher.  When you compile the v1.2.0 firmware, ensure you select the 1M(256K SPIFFS) setting or there isn't enough room for OTA.

I have a nodeMCU which I use for my prototyping and it seems to work fine.  Let me know how you go.

potentstudios, 2017-03-15:
The firmware is working flawlessly! The red light just scared me a bit. The
red light goes away after I set up the settings in the Web GUI and then it
briefly is on while turning on the device. Other than that it works
flawlessly with Jinx! software and I couldn't be happier! One thing I have
noticed is there's a bit of lag but I can work around it for my
uses because I'm only using midi information from Reaper to control the
lights so I put the triggers a little early in the timeline so it's on
beat. Thank you so much for your help! I'm so glad I found your tutorial
and that you were gracious enough to answer all my questions! Thank you!!

-----------------------
*Lee McIntosh*
Web / Graphic Designer
Portfoleeo.com <http://portfoleeo.com/>

<http://i.imgur.com/Yeg7t.png?1> <http://i.imgur.com/Yeg7t.png?1>

mtongnz, 2017-03-27:
I'm glad you got everything working.  The source will be released soon so hopefully the lag can be minimized by the community :)

closed 2017-03-27 by mtongnz