mtongnz/ESP8266_ArtNetNode_v2 #47: Artnet goes offline

sqmsmu, 2017-04-08:
Artnet will go offline during a show and wont return online until i reset esp8266..

sqmsmu, 2017-04-08:
i am using nodemcu with wemos file.

mtongnz, 2017-04-10:
Could you give more details of the circumstances of the reset.  There's not a lot to go on here.

DMX, RDM, WS2812 mode?  What console you're using, broadcast or unicast data.... maybe some screenshots of the Web UI settings pages

sqmsmu, 2017-04-10:
https://drive.google.com/file/d/0B1tHnFA_0UOSalFjeWcwU245aW8/view?usp=sharing
https://drive.google.com/file/d/0B1tHnFA_0UOSMGwtM0U4U2lYRzA/view?usp=sharing
https://drive.google.com/file/d/0B1tHnFA_0UOSQmpULUl6c19Cazg/view?usp=sharing
https://drive.google.com/file/d/0B1tHnFA_0UOSdTJ2b0sxbkJtNjA/view?usp=sharing

sqmsmu, 2017-04-10:
i am using all default settings...and i am sending dmx through madmapper on mac. everything works fine, but after like 5-10 min artnet node goes offline and then comes back online after few seconds.. As i will be using it for a light show with around 30 universes, i cant afford these resets.

sqmsmu, 2017-04-10:
Although when it goes offline, the Web UI shows it is ON but MadMapper says it is offline.

mtongnz, 2017-04-10:
This is a known issue - see here: #41 

This firmware is still beta - I wouldn't be using it in a show.  Try firmware v1.2 available [here](https://github.com/mtongnz/ESP8266_ArtNetNode_DMX)

closed 2017-04-19 by mtongnz