mtongnz/ESP8266_ArtNetNode_v2 #48: DMX IN

Lazerjules, 2017-04-10:
Hi Mtongnz

Fantastic project, excellent work!

Have you implemented DMX IN yet? I don't seem to be able to get it to work. 
DMX out on A and B no problem, working perfectly. 

Using DMX IN, I am assuming I should see the input values in DMX Workshop? Checking the node list I see what's in the attached picture so something is happening. 

Any advice would be much appreciated thank you. 

[image]

[image]

mtongnz, 2017-04-10:
Ensure you set the DMX In broadcast address in the Web UI.  It should be working.  I'll check it again when I get a chance.

Lazerjules, 2017-04-10:
Hi Matt
Perfect thanks. Operator error! I hadn't set the broadcast address! Apologies.

closed 2017-04-10 by Lazerjules