mtongnz/ESP8266_ArtNetNode_v2 #55: Schematics differences with v1?

sinedied, 2017-05-31:
First, thanks for your project and sharing the instructions with it! 👍 

I successfully made a simpler version based on your v1 schematics, using a nodemcu 1.0 board (with 3.3V everywhere) and a single dmx output. It works very well 😄 

I looked into the v2 schematics, and saw that you added a bunch of resistors, could your explain why? If I only use it for DMX output, not pixels/RGB is it useful to move to the new design?

Thanks!

mtongnz, 2017-06-05:
The resistors are for DMX termination and are recommended for RDM.  I based my design on an Arduino shield design by Mat Hertel - he explains it here:  http://www.mathertel.de/Arduino/DMXShield.aspx

sinedied, 2017-06-05:
Thanks for the info!

closed 2017-06-05 by sinedied