mtongnz/ESP8266_ArtNetNode_v2 #58: Can't compile latest master, project refering to another .ino

VideoRipper, 2017-06-19:
It's impossible to build the latest version of **espArtnetNode_2.0.0_b5g.ino** since the functions:
- wifiStart()
- webStart()
- artStart()
- portSetup()
- doNodeReport()
are in **startFunctions.ino** (which should be a separare sketch by looking at the extension) and not included or declared in **espArtnetNode_2.0.0_b5g.ino**.
(This also is true for **ajax.ino** and **firmUpdate.ino**)

mtongnz, 2017-06-22:
I split the main sketch into 4 files to make it easier to read - it was too long as one.  They should all open when you open the main b5g.ino file as tabs in the arduino IDE.

If this doesn't happen, there's a little down arrow on the right side of the tabs pane.  Click it and a drop down should appear with the option to add a New Tab - click that and then enter the file names (ajax.ino, firmUpdate.ino & startFunctions.ino).  This should add the files as new tabs and then it should all compile correctly.  After you save, it should keep the tabs the next time you open.

The other option is to copy everything into the main sketch.

Let me know how you get on.

VideoRipper, 2017-06-22:
We've got it going by merging all project (*.ino) files into one, but it's not ideal: the code should be split up in separate *.cpp/*.h files.

closed 2017-06-22 by VideoRipper