mtongnz/ESP8266_ArtNetNode_v2 #61: Stand Alone mode not working

Vrejf, 2017-09-07:
Hi
I can't get the Stand Alone-mode working in v2.0.0 (beta 5g)
When connected to an existing wifi everything is working correctly.
In Stand Alone-mode, i can connect to the wifi, and the  status on Port A says "DMX output".
But when trying to control the fixture nothing happens.
Am I doing something wrong or do you think its a bug?
And I can not find Style tab I read about.

Thank you very much!

ScholliYT, 2017-09-17:
If I'm right you cant send data to it in Stand Alone. Only play saved scenes.

from espArtnetNode_2.0.0_b5g.ino line 10

> <p class=right>In normal mode, the hotspot will start after <i>delay</i> seconds if the main WiFi won't connect. If no users connect, the device will reset and attempt the main WiFi again. This feature is purely for changing settings and ArtNet data is ignored.<p class=right>Stand alone mode disables the primary WiFi connection and allows ArtNet data to be received via the hotspot connection.

mtongnz, 2017-09-18:
It should work.  Ensure that the stand alone mode box is checked.  Check your console/computer has the correct IP and subnet and that it is broadcasting/unicasting to the correct IP.  By default, the IP's for stand alone should be in the 2.0.0.0 range with a subnet of 255.0.0.0 and your control software should unicast to the ESPs IP or broadcast to 2.0.0.255 or 2.255.255.255

mtongnz, 2017-10-03:
@Vrejf Could you please confirm if you got this going or not.  If you have, please close this issue.  If not, try giving us a bit of info on your setup (what software, IP settings on the computer...)
Thanks

mtongnz, 2017-10-10:
Haven't heard back with confirmation of this issue so I'm assuming you've got it going.  Closing this issue

closed 2017-10-10 by mtongnz

bombcheck, 2018-03-05:
Just tried stand-alone-mode: Attached wifi-usb-dongle to my pc, connected to the hotspot and activated hot spot mode: Totally freeze/crash: I can connect to the hot spot, but neither the web ui nor my led pixel are responding any more. Had to re-flash with data-wipe.