mtongnz/ESP8266_ArtNetNode_v2 #62: Options resetting frequently

T3rry12, 2017-09-20:
Sorry i wont be much help as most of this is WAYY over my head, but i did get this up and running with a few hiccups along the way, one thing i am noticing though is, first off the reboot from the web interface doesn't seem to work, but second, almost every time i reboot the wemos, it loses all the settings, i have to ad-hoc to it to re-set up the wifi, but then when i reboot the wemos again, most of the time, it loses the settings again, a few times it has retained them, but not often...

i don't know if this is related to bug #41 or not.

ammuller, 2017-09-26:
Please try https://github.com/ammuller/ESP8266_ArtNetNode_v2
It is preforming relatively stable with WS2812, keeps settings no WDT issues. 
The lights get stuck every few minutes but can be reset with web interface.

mtongnz, 2017-10-03:
Thanks for the contribution @ammuller I am looking forward to trying your code when I get a bit of time.