mtongnz/ESP8266_ArtNetNode_v2 #65: Simple PWM output

amcdawes, 2017-09-25:
First off, this is an awesome tool and thank you for sharing it! I feel a bit silly raising an issue, but I'd suggest a cool feature that would be another output style of "simple PWM output". A specific pin run in PWM mode tied to a DMX channel would enable all sorts of clever uses (servo, a single simple LED, driving solid state relays to control big stuff on stage). I can think of half dozen uses in shows I've helped with over the years. I'll be using the LED-strip driver mode _a_lot_ but would use a basic PWM output even more. I'm happy to help implement something like this, but I haven't jumped far enough into the code to see how to do so. I have my own clunky script for artnet DMX to PWM but it's missing all the cool RDM, web UI and general awesomeness of what you have here.

mtongnz, 2017-10-03:
This shouldn't be too hard to implement.  Look at adding it in the DMX handling function.

Unfortunately, I haven't had much time to play with this code for quite a while and work is only getting busier.  Please do share when you've got some code though.

alienfive, 2017-11-25:
I just tried the ARtnet to DMX and it work's great! 
Well done and cool project!

Since i'm modifying some 50W RGB-flooder to "DMX/ ART Net" it would be nice to have at least min. 3 PWM channels for each color one.
Since I'm not a super programer, I started to go through the code, but at the moment I'm not really familiar with it and have no idea how to start ;(
Maybe you have some ideas & hints?

thejacko12354, 2017-12-14:
OT!
@alienfive could you share how you did modify the flooder? I'm really interestet.

cpethica, 2018-01-16:
Amazing software, thanks for taking the time to develop this mtongnz. I needed a DMX to PWM feature for project so I've tweaked your code a bit and it seems to be working fine. I've only used 4 channels so far but I'm sure you could add more - code is here, line 348 for the relevant code:

https://github.com/cpethica/We-Are-Symphony/blob/master/espArtnetNode_PWM/espArtnetNode_PWM.ino

I have it running on WEMOS D1 mini - you need to be a bit careful about which pins you use. As a result I've commented out all the status LED pin code as this was interfering with things and I was running out of pins to use... My coding is pretty bad but hopefully this will be useful to someone.

jfs-de, 2018-01-17:
Hello Matthew, Friends, 
I changed the 1.20 code of to outputting 3 PWM 1024Bit Values. Threw out everyting not needed ( DMX dependency, etc... ) Gamma correction included.
AP Password: !)&&JochenSchefe1

Have a look if You want 

[ESP8266_ArtNetNode_DMX-120-b-pwm.zip](https://github.com/mtongnz/ESP8266_ArtNetNode_v2/files/1639245/ESP8266_ArtNetNode_DMX-120-b-pwm.zip)

alienfive, 2018-02-09:
Hello together,

Thanks @jfs-de  & @cpethica for providing your solutions!
At the moment i just had time to test the one from jfs-de and it works great!

Maybe it should be implemented also in the artnetnode !?

@thejacko12354  i will update the RGB-Flooder with 3x TPS61500 (constant current step up regulatr from TI) --> this ASIC is controlled with the PWM signal of the ESP ;)

scamiv, 2018-03-21:
Here's an alternate solution, DMX_TX_A/B are now on pin 5/4 but you can change that to your liking, i disabled the directional pins as those aren't needed by me.
The array pwmports defines alll gpio's which will be exposed to artnet. {16, 0, 2, 14, 12, 13, 15, 3, 1};

Those ports will be availible at universe 10, for net and subnet the values of portA will be used.
https://gist.github.com/scamiv/687dfa8e80645878399900fc2c694ee0

tohox, 2018-05-09:
@scamiv Nice work adding the PWM outputs and fixing some of the earlier version's problems!

I've tried your 2.0.0d (beta 5g) version on a Wemos D1 mini Pro and found that only PWM outputs appear to work at the moment. As soon as I send anything to a universe associated with the PORTA DMX output port the ESP freezes. I've tried changing the pins associated with the DMX ports back to what worked with @mtongnz  original version but it seems to have no effect. The Wemos still crashes as soon as I send something through the DMX out port.

Also, in order for your sketch to work I have to select the "Wemos D1 and R2 mini" board and specify 4M (3M spiffs) instead of the "Wemos D1 mini pro" with 16M (15M spiffs).

Any ideas?

Thanks,

vwvanfan, 2018-10-06:
Firstly thank you for this code. I've got it running and it's controlling my strips nicely via WS2812. 
I've since tried all 3 of the offerings above to get PWM working instead of WS2812 or DMX, but i'm struggling. It's most definitely my fault - my knowledge is poor. The closest i've got is @jfs-de 's version, which I actually managed to get on to the Wemos (couldn't succesfully compile the other 2 - my fault for being dumb I know). My issue with @jfs-de 's version though is that I just can't join the access point. I'm entering the password you list "!)&&JochenSchefe1" but it just won't have it. Have tried uploading the code via arduino ide and also the firmware update method with the .bin but neither gets me any further. 

I'm building some artnet connected lanterns (chinese paper lanterns) so i need all the pixels to be the same colour. I have 2 options - use WS2812 and modify the code so every pixel has the colour of pixel 1, which has its' address set in the usual way via the browser interface. Or get PWM working. 

If i'm honest, I think I prefer the WS2812 version as it involves literally no external hardware, just the Wemos, a power supply (battery charging board) and the strip. Not against using PWM though if I can just get some of the offerings above working. 
Could anybody help with either some modified WS2812 code or some assistance with PWM. A .bin that I can just update the Wemos with via @mtongnz 's firmware update utility in the browser would be incredible! 
Thank you so much!

cpethica, 2018-10-08:
@vwvanfan 

I have a PWM version that worked for me. I’ve added a binary here:  http://github.com/cpethica/We-Are-Symphony/tree/master/espArtnetNode_PWM

Been a while since I looked at this but if I can find a Wemos today I’ll test again.

If you use WS2812 why do you need to change the code to get all the lanterns the same colour? Can you not do this at the control end with your lighting software?