mtongnz/ESP8266_ArtNetNode_v2 #67: 144 LED Strips

xOr23, 2017-10-05:
Hello, I use 4 led strips with 144 led/strip snakeline and as a program Jinx. If I only use 144 leds everything works very well. As soon as I patched the second 144 leds, I get an offset of 26 leds between the first and second 144 leds. I have tried out all possible output patches, but unfortunately without success.

djpeterlewis, 2017-10-05:
Remember 170 Pixels is 1 universe , so your mapping after the first 144 pixels is incorrect.
Is this one long string ? or are you using 2 outputs ?

djpeterlewis, 2017-10-05:
First 144 pixels starts at universe 0 , address 0  then address after your mapping of 432, it should then start at Universe 1 addresss 0

mtongnz, 2017-10-10:
As @djpeterlewis stated above.  Closing this issue

closed 2017-10-10 by mtongnz