mtongnz/ESP8266_ArtNetNode_v2 #68: 'rdm_data' has no member named 'clear'

ririfonfon, 2017-10-13:
[image]

Hello
when I check, I find myself with this problem.
do you have an idea of where it comes from?
thank you.

gravicapa19, 2017-10-24:
Hey.
I have the same problem, only on the windows machine.
On a Linux machine, the compilation runs without errors

zanglerecke, 2017-11-17:
the real function is included in file libs/espDMX_RDM/rdmDataTypes.h 

this is the same file/definition as in the espArtnetRDM.. there the header is added, but the function clear is missing... -> add the function and it will be done ;)

sent a pull reques.