mtongnz/ESP8266_ArtNetNode_v2 #70: MQTT support

donnib, 2017-11-27:
Hi,
I am planning to mount roughly 24meters of LED lights in the ceiling of a big room that's dual purpose, one for parties and one as office therefore i was thinking to use this soft to control them with artNet from my Mac for parties but otherwise i need to control them as normal light (to light up the room) so i was thinking that adding MQTT support inside this would let me control the brightness/color from my HomeAssistant system leading me to control the lights from a wall switch with Zigbee support. Has anybody thought about that? I mean the MQTT support and artNet could work at the same time. It's just who is last that controls them that wins, or you could change mode in the WebUI. What do you think ?

The only problem i see is that i hope there is enough space in the ESP8266 to add MQTT support side by side with artNet.

/donnib

ammuller, 2017-12-04:
What functions would you require over Mqtt?

lukas2511, 2017-12-05:
Being able to set channel states and reporting state on changes would be great.

There are a ton of MQTT brokers and libraries, and there is example code for basically every programming language, so this would make a great API to easily connect to an ArtNet or a physical DMX installation with custom code.

Also it would make it fairly easy to integrate into existing home automation software like openHAB and Home Assistant, as existing plugins could (with special configuration or minor modifications) just work.

But I have to say that I'm not sure if this might be a bit too much for one piece of software...
I like features, but I also like devices that have just one job and are working rock-solid.

ammuller, 2017-12-06:
Would you be intrested in a Version with only Mqtt and ws2812 as a start?

Giaitzoglou, 2017-12-29:
@anmuller Hi @anmuller do you have the mqtt version I just want to dimmer some lights using a dmx dimmer not any scenes just dim and brithness!
Any help will be appreciated 
Thanks 
Chris

Giaitzoglou, 2017-12-29:
@ammuller sorry autocorrect

mtongnz, 2018-01-23:
MQTT isn't a feature I plan on implementing.  I'm sure there are other projects out there which use this.   Feel free to take my code and make something new with it but I wouldn't bother with Artnet if you're using MQTT for commands

closed 2018-01-23 by mtongnz