mtongnz/ESP8266_ArtNetNode_v2 #71: According to the version and schema

vormy, 2017-12-05:
Hello. Would like to repeat your project to control light by DMX
Tell me if I use the scheme
[image]
I need to use the firmware 
espArtnetNode_2.0.0_b5g.bin

If I use a different scheme   
[image]
Then the firmware should take another
ESP8266_ArtNetNode_DMX_1.2.0.bin	

Libraries are the same for both projects?

ElToberino, 2017-12-07:
As far is I see, there is no difference between the layouts concerning the software version.
The newer layout uses octocouplers and DC-DC converters; this is a little safer, but has nothing to do with software functionality. I use the newest software version with the former layout.

vormy, 2017-12-08:
but what about the LEDs? firmware different indication. With the new firmware v2 on the old scheme will not work LEDs?

ElToberino, 2017-12-09:
You're right. LED indication is different. "espArtnetNode_2.0.0_b5g.bin" uses only one ESP-Pin for a digital output to drive three APA106-LEDs as indicators. The former software versions used two ESP-Pins, each for a standard LED.
I didn't focus on that because I don't need the indication LEDs and so I leave GPIO12 and GPIO13 unconnected.

mtongnz, 2018-01-23:
The libraries are different on the V2 software.  I wrote the Artnet and DMX libraries from scratch for V2 to allow RDM functionality.
In terms of hardware, there are a couple of major changes.  As I added RDM, I needed to add a direction pin and the RX pins to allow receiving data.  I decided to change from single LEDs to APA106 to reduce the pin count and allow for full RGB status LEDs.  The final change, as stated above, is the addition of electrical and optical isolation of the DMX outputs - this isn't required but makes for a safer system if there is an issue (such as power spike) on one of the outputs.
In theory, the V2 software should work on the V1 hardware but as I no longer have any V1 hardware, I can't test this.  If you're building new hardware, I'd recommend using the new schema as your base design.

closed 2018-01-23 by mtongnz