mtongnz/ESP8266_ArtNetNode_v2 #74: nodeMCU not saving changes from web GUI.

potentstudios, 2017-12-14:
I've tried several times to change the settings and it seems to be finicky on what it saves. I'm using a setup of 10 MCUs and some have no issues while others won't save. Any thoughts?

lampenschakelen, 2018-01-28:
Hi potentstudios , 

You can try to erase the esp8266. 

This fix the problem for me. 

http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

potentstudios, 2018-01-28:
Thanks I'll try that!

bombcheck, 2018-03-12:
Maybe this part of the setup routine?
```
// Check if SPIFFS formatted
  if (SPIFFS.exists("/formatted.txt")) {
    SPIFFS.format();
    
    File f = SPIFFS.open("/formatted.txt", "w");
    f.print("Formatted");
    f.close();
  }
```

It should be more like this right?
```
// Check if SPIFFS formatted
  if (!SPIFFS.exists("/formatted.txt")) {
    SPIFFS.format();
    
    File f = SPIFFS.open("/formatted.txt", "w");
    f.print("Formatted");
    f.close();
  }
```

khastings1066, 2018-06-18:
The solution proposed by bombcheck worked for me with beta5g.
Currently SPIFFS is wiped every time the system restarts.

potentstudios, 2018-06-27:
Has anyone else tested/confirmed bombcheck's fix?