mtongnz/ESP8266_ArtNetNode_v2 #76: Weirdness with pixels and Jinx

ke4pjw, 2017-12-18:
I have found that I can configure my WS2811 strip and WS2812b pixels to work, only if I choose specific numbers of pixels to control. Otherwise it just gets weird. Sometimes the whole strip will flash like a rainbow. It will change colors when I change intensity. I have not put my probe on it to view the output. If I choose a certain number of pixels in the config, it will work correctly, but that number appears to be arbitrary. 

I assume DMX channel 1 is the address of the first pixel when running in WS2812 mode.

BTW, wonderful project Matt! I was able to go from 0 to pixel control in about an hour. Very nice! Never touched the ESP8266 before this.

PS: I am running on the B port and both ports are configured for WS2812. The problem appears on both ports.

juli303, 2018-12-21:
I had the same problem. I'm no expert but apparently but apparently this happens because the signal voltage is only 3.3Vs so the LEDs sometimes don't register a HIGH signal as HIGH. Some people solved it by adding a capacitor between +5V and Ground and it works for me too.
Hope this solves your problem

madis25, 2018-12-21:
Guys... can you stop arguing about this and start to be clever (move your braincells or whatever).
I'm on the way to make this project, and have already fix idea how to solve this problem.
Shortly, start from power source. I connect ESP to the end of LED-chain. There's will be voltage drop anyway. Second, read spec of ESP and 8212, there's written that low output may be higher than real 0V. So, just add couple of resistors and 3,3V zener and that's all! Just moving real 0v little higher. Of course, calculation first, then testing. If the first 8212 LED will reciving information perfectly, ther everything else will work perfectly until the end of chain.
[image]