mtongnz/ESP8266_ArtNetNode_v2 #77: Help with ip

Giaitzoglou, 2017-12-23:
Hi is it possible to tell if the art net esp can work in 192.168.1.0/24 networks ?
Or only 2.0.0.1 networks and also in you new firmware you have at ports  
Net and subnet can you please tell me how to configure it correctly ?
Thanks in advance 
Chris

Giaitzoglou, 2017-12-23:
Also what I want is to just control my Dmx dimmer lights is there an easer way with just mqtt and dmx ?

hsiboy, 2017-12-26:
Hi,

You may want to reconsider the 2.0.0.0/8 address that the ESP8266 uses for its own IP when running in "hotspot" mode.

1. This IP range is routable over the public internet.
2. Many security products will block this IP range.

Please consider using a non public internet routable, private IP address range.

Giaitzoglou, 2017-12-29:
So you say that the Artnet protocol will work in a 192.168.1.0/24 network . I want to use node red to control an dmx dimmer but I can’t make it work ! And I thought because the network is that and in the art net node I used the up of the dmx esp say for example 192.168.1.28.

Giaitzoglou, 2017-12-29:
@hsiboy

mtongnz, 2018-01-23:
@hsiboy The 2.0.0.0 (or 10.0.0.0) range is specified by the Artnet standard.  The network shouldn't really be connected to the internet and in stand alone hot spot mode, it is impossible to connect to the internet anyway.  I'll be leaving this as is but feel free to change it if you want.\

@Giaitzoglou The IP can be changed in the web gui to anything you want.  Provided you have all your devices in the same range and there's nothing blocking your packets, Artnet should work in any IP range.

closed 2018-01-23 by mtongnz

hsiboy, 2018-01-24:
@mtongnz yeah, i think the 2.0.0.0 became the standard back when that IP address wasn't publicly routed on the internet and the 10.0.0.0 network has always been designated private, so that's a good choice.

But, its your project 👍

mtongnz, 2018-01-25:
@hsiboy The reason for using the 2.0.0.0 range in the Artnet protocol is simple - each device manufactured should come with its own unique IP.  From the official Artnet spec:

> By default an Art-Net product will factory start using a Class A IP address scheme.   This allows Art-Net products to communicate directly and without the need for
a DHCP server to be connected to the network.

> The use of Class A addressing is allowed within a closed network. It is important to
ensure that Art-Net data is not routed onto the Internet.
Products implementing Art-Net should default to the Primary IP address of 2.?.?.?.

Using a range such as 192.168.0.x would limit this to 255 devices which isn't very many.  Read more Artnet here: https://en.wikipedia.org/wiki/Art-Net and here: https://artisticlicence.com/WebSiteMaster/User%20Guides/art-net.pdf

I've tried to follow the Artnet spec as closely as possible - hence the 2.0.0.0 range.

As for this range being routed over the internet, this isn't actually possible in hotspot mode as it has no wan connection.  In my opinion, a lighting network with Artnet running on it should be separated from a network with internet connectivity anyway, either by using separate switches or implementing vlans.

And finally, the IP can be changed via the web ui.  When connecting to wifi (not hotspot mode), the device defaults to using DHCP and so will get whatever IP is assigned by your switch/router.  Once again, you can override this in the web ui and set a static IP of your choosing.

You are more than welcome to change the default IP in the source config if so desired but I'd prefer to follow the Artnet spec on this and so wont be changing it.  In theory, I should give each device a unique IP based on my OEM code & an arbitrary device ID, but this would mean connecting to the web ui would be a pain as the IP would be different on every device.

I hope this clarifies my stance.