mtongnz/ESP8266_ArtNetNode_v2 #78: Will this work with the ESP-32 chip

Capii1, 2017-12-30:
I have the beta5g running on an ESP-12E chip and it works just fine. (When using the precompiled bin files) Will it be ported over to the ESP-32 chip in the future and also have you figured out why the source code does not work when compiled. I've tried to make some minor adjustments to the source however when I compile the unedited and edited source files it does not output anything to the LEDs.

idtres, 2018-01-21:
hey you already ported all to esp32 ? There is another UART on esp32... its possible a 3 out artnet node ?
https://www.youtube.com/watch?v=GwShqW39jlE
plus there are several pin on esp32 por pwm

tinic, 2019-04-08:
It's totally possible to port this to esp32. I did a quick and ugly port here https://github.com/tinic/ESP8266_ArtNetNode_v2  for a ESP32-POE  module.

It'll take another few weeks to port over all features and clean up the code base but the original code base makes a great starting point. I found a few obvious bugs which will likely cause random crashers. One benefit of running recent gcc compilers is that they guard against that. Also, instead of using assembly to drive GPIOs, using SPI yields way better timing for WS2812 LEDs without any flicker.

LoneWalkerWolf, 2021-03-12:
Yeah, @tinic is totally right. 
We just need to change some lines of code in the libraries, and BINGO, it's ESP32 capable.

idtres, 2021-03-12:
How many DMX outputs can ESP32 handle ? this will be a nice feature

LoneWalkerWolf, 2021-03-12:
Honestly, I have no idea. It has a tonne more pins, but I haven't tried any sort of DMX software on it yet. 
I'll let you know how it goes when I try it out.

idtres, 2021-03-12:
btw check this out
https://hackaday.com/2019/05/07/lots-of-blinky-esp32-drives-20000-ws2812-leds/

DachsbauTV, 2021-03-12:
This would acutally be amazing because it would be one step closer to having support for the amazing M5StickCPlus, which would make an amazing wireless Art-Net receiver

idtres, 2021-03-12:
At the moment I am working with the WS5500 ethernet adapter + esp32, and it quite easy to do UDP packages...
It would be a nice alternative to avoid bottleneck and WiFi errors for critical applications like DMX ( well, its not reall critical, but its time sensitive ).
There should be a alternative to the DMX output library of the ESP32 to have more than 1 single output

LoneWalkerWolf, 2021-03-12:
Yeah, sounds good.
I'm trying one with the ENC20-something. I'll let you know how it goes.