mtongnz/ESP8266_ArtNetNode_v2 #82: Node freezes after several hours

idtres, 2018-01-20:
Hi, Im using a WemosD1, one output set to dmx another set to PixelStrips, and also a 3 strips for the status LED
It works amazing, super fast, no delay or flickering, but after several hours the node stops sending artnet data and wstrips data.
1 status LED goes Red, and the other 2 stop responding, but at the same time, I can ping the device and the webpage responds

not sure what the RED led means.

Im sending data using VVVV, I slowed down the output to 10 frames per seconds per universe, still same problem.

Please let me know what other test I can try to debug this.

I tested sending 1 universe only, (dmx) and it worked just fine for  5 hours.

querex007, 2018-01-21:
I guess this is a known issue of the project...

I found a quick work around:
When the node freezes (and still responding to ping and web GUI), you can send 2 reboot CURL's, which makes the node operating again...

CURL: -X POST -H "Content-Type: application/json" -d "@rebootAmbilight.txt" http://192.168.x.x/ajax

As a reboot only takes 2 or 3 seconds, you can schedule this curl on your notebook / computer every 15 minutes.

terrag42, 2018-02-19:
Is there a way to use one ESP8266 to do nothing but periodically issue curl commands to my other ESP units which are running?

idtres, 2018-06-17:
I have been using the node with 2 dmx output receiving data from Resolume for almost 4 hours
The only changes I made are feeding the watchdog on the main loop ( ESP.wdtFeed(); )and commenting all lines about neopixel for the status leds,,, for some reason ( well its more like a feeling, no tech reason ) there is something wrong with that. The node as DMX is way more stable than for Neopixels. Using 600 pixels on 1 output only, it freezes all the time in less than 15 minutes. Same node as dmx can last 2 hours with no problem. After disabling all the status neopixels declarations, I was able to run the node for 4 hours without any problem. 

I wrote a script to log if the ping stops, but this did not work, may times the node did not reply ping,but still works just fine. I was about to write a vvvv patch to evaluate 2 nodes, one as input a second one as output, but I wont have any way to tell witch one fails, so I will be using dmx as artnet sender + a webcam + rgb fixture, to change color every 1 second, and evaluate using the webcam if the color is correct. 

@mtongnz did you found any other tip regarding this ?