mtongnz/ESP8266_ArtNetNode_v2 #84: Description of status LEDs

meelow, 2018-02-11:
Hi Mathew, 

thanks for sharing this awesome project. I do want to use your code to upgrade my own transportable lighting project from using OSC as input to go for DMX. 

I read and saw in the code that you now use the first the LEDs of the attached stripe to show the status of the ArtNetNode.  
From my experience (having some problems getting everything up and running - probably mainly because of old EEPROM data) I could see that:
the third LED seems to be the 'hearbeat' (blinking green in my case)
The first two leds seem to reflect PortA and PortB status (if yellow) OR
show (WiFi?) connection problems (if red). 

Could you either describe your logic between colors / LEDs or point out where to look in the code to understand what errors pop up? 

That would be a very good addition to the (pretty good) readme.

Let me know if i can support in writing a 'status-led' guide... 

Thanks 

Milo

mtongnz, 2018-02-17:
I haven't put much thought into the status LEDs yet.  It was more a proof of concept prior to me designing and ordering the PCBs.  I plan to make it much easier to set desired colours for each status.

Here's what it is for now:
Status LED:
    PINK - startup
    GREEN flashing - heartbeat
    RED flashing - nodeError isn't empty. generally caused by a WDT reset (the string will send over artnet)

Port A&B LEDs:
    CYAN - DMX input received and sent to artnet
    GREEN - DMX received from artnet - port in WS2812 mode
    BLUE - DMX received from artnet - port in DMX or RDM mode
    
The colours can be tuned using the definitions at the top of the sketch.  I haven't really thought through this much so feel free to come up with some better ideas and share them here.  I also haven't tested using WS2812 status LEDs yet but it should work fine.

markusb, 2018-07-14:
Just a comment, I'just starting to play with it on a Wemos D1. Things are partially working (DMX yes, WS2812 no). I found the status LEDs a great help, as they show me that I have my WS2812 LEDs wired correctly. The status LEDs are a great feature to see what is working and what is not.

Just one question: It looks like the 1st three LEDs on port A are used for status. So am I supposed to have three additional LEDS on port A for status or with the status vanish, when I start to use the string for lighting ?