mtongnz/ESP8266_ArtNetNode_v2 #85: 3800 Pixel LED installation

terrag42, 2018-02-13:
I am building an LED installation using 3800 pixels. Are four ESP8266s suitable for such a task using your software? I've attached my proposed LED layout here:
[LED Tree Proposed Layout.pdf](https://github.com/mtongnz/ESP8266_ArtNetNode_v2/files/1721124/LED.Tree.Proposed.Layout.pdf)

mtongnz, 2018-02-17:
Yes and no.  It is definitely capable of doing this but my firmware is still a bit buggy.  It has a habit of locking up after running for 10-15 minutes - see #82 .  I am working on a fix and that is a work around for now.  No idea when/if this bug will get fixed as I have limited time for this project for now.

terrag42, 2018-02-19:
Thanks for the quick answer. I saw your answer from issue #82 "As a reboot only takes 2 or 3 seconds, you can schedule this curl on your notebook / computer every 15 minutes."
Is there a way to use another ESP8266 to do nothing but periodically issue these commands to my other four ESP units?

mtongnz, 2018-02-20:
That would definitely be doable and shouldn't be that hard but I'm not sure on the actual code that would be needed.  I'd say there's code somewhere on the net for constantly updating remote servers with data (such as sensors) which could be easily modified to do this.

bombcheck, 2018-02-21:
My personal experience is, that using wifi is quite unrealiable. On places, where many many active wifi networks are found (on a big Fair for example), it is quite hard to even get a few bytes through a wifi channel. And using many ESPs with many universes will produce a large UDP-flood which could even break down a "normal" network at home.

When reliability is a concern, I would prefer a solution with wired network, if possible.