mtongnz/ESP8266_ArtNetNode_v2 #86: Web GUI not reachable

user-51, 2018-02-22:
Hello!

I flashed the v2 binary on a NodeMCU. Connection to the wifi hotspot works, but i cannot access the config website.

curl gives the following output:
```
curl -vvv 2.0.0.1:80                                                                                                                                                                            
* Rebuilt URL to: 2.0.0.1:80/                                                                                                                                                                        
*   Trying 2.0.0.1...                                                                                                                                                                                
* TCP_NODELAY set                                                                                                                                                                                    
* Connected to 2.0.0.1 (2.0.0.1) port 80 (#0)                                                                                                                                                        > GET / HTTP/1.1                                                                                               
> Host: 2.0.0.1                                                                    |                               
> User-Agent: curl/7.58.0                                            
> Accept: */*                                                                                    
>                                                                                                                                                
< HTTP/1.1 200 OK                                                                                                                             
< Content-Type: text/html                                                                                                                                                     
< Connection: close                                                                                                                                                                                  
< Content-Length: 17057                                                                                                                                                                      
< Connection: close   

```
Erasing the flash with esptool before flashing the firmware did not have any effect.

futuristicblanket, 2018-02-25:
Hi user-51,

I a, continuing to have the exact same problem. I have however found a really strange workout. When accessing the web GUI using either Chrome, Firefox or Safari from macOS 10.10.6 my ESP8266 completely freezes up and requires a power cycle; however, when accessing the web GUI from Chrome on my Android Lollipop phone everything works as expected. Artnet from my laptop as no effect, just the web gui. Perhaps try accessing the web gui from your phone?

Hope that helps, Thomas

futuristicblanket, 2018-02-25:
Update:

I also tested in Chromium on Ubuntu 17.10 and it was able to access the webpage as expected. Still no luck on macOS. Strange.

mtongnz, 2018-03-25:
I haven't experienced this issue ever.  I'm not sure what the cause could be.

@scamiv has found and fixed a few issues [here](https://github.com/mtongnz/ESP8266_ArtNetNode_v2/pull/92) and one is a memleak in the way I am serving the web GUI.  I have yet to test it but it looks promising and I'd recommend giving it a try.

futuristicblanket, 2018-03-29:
Thanks @mtongnz. I can confirm that in @scamiv 's version the web GUI works as expected on MacOS. Thanks, Thomas