mtongnz/ESP8266_ArtNetNode_v2 #88: Cannot connect anmore to hotspot.

salvonos, 2018-03-05:
I have uploaded your firmware fine on my nodemcu fine. It was everything working. I have changed the wifi setting and now I cannot connect to it anymore. The Hotspot does not appear on my list of connections anymore. I have tried to upload the firmware again but nothing happens. 
What should I do?

bombcheck, 2018-03-05:
Maybe try to flash the firmware again but wipe flash before. You can use this flasher to do so:
https://github.com/marcelstoer/nodemcu-pyflasher

mtongnz, 2018-03-25:
If you've set the wifi details, has the ESP connected to your home wifi?  The hotspot only shows up if the device is set to stand alone mode or if it can't find/connect to the wifi.