mtongnz/ESP8266_ArtNetNode_v2 #90: Source code and Wemos

nll41, 2018-03-05:
Hello
I have flashed a Wemos D1 with the wemos .bin file, and i works perfectly.
But I can't make it to work when I compile the source code myself.
Has anyone tried with succes?

best regards
Niels

thomsenle, 2018-03-11:
i have the same problem :( the compiled sourcecode does not work on the wemos

thomsenle, 2018-03-12:
Yes, have changed the ports according to my setup. Want to use txd2/gpio4/d2 as output for a single line of pixels. Enabled  "esp01 mode" and changed  pinout to my setup. I played arround a little  last evening and all that happened was that the strips turned on at full white. Have to take a deeper look this evening with osciloscope.

bombcheck, 2018-03-12:
Hm... The only time I had a full white matrix was when I compiled with 80 MHz instead of 160 MHz.

bombcheck, 2018-03-12:
TX on Wemos should be GPIO 1 in the code.

bombcheck, 2018-03-12:
Can you give us the exact pinout you used in your code?

thomsenle, 2018-03-12:
will do this evening... in the office right now. 

Grüsse aus Leipzig

nll41, 2018-03-12:
Hello
Thanks for the advice, I changed to 160MHz and recompiled, now it works :-)
best regards Niels

bombcheck, 2018-03-13:
Great to hear it is working now.

bombcheck, 2018-03-13:
So I think this one can be closed then. :)

closed 2018-03-13 by nll41