mtongnz/ESP8266_ArtNetNode_v2 #95: Wifi not conecting and other Errors

GoMini-01, 2018-06-09:
My esp won't conect to my local WiFi anymore (it workt at one point but decidet to stop now).
The WiFi setings are corect since it workt earlyer...

With the V1 version which workt flawles previously (first install). Now it conects to wifi and works "fine" but the A port won't work only B runs corect. At one point i got it to run both ports but it the output flickert.

edit: got it working ... (but for how long...) after uptating from v1 to v2 by web ui (the other ways didnt work for me ) it did not want to accept the wifi setings and startet in hotspotmode ... anter completly powering it of and on it worked...

Hardware: 
  AI-Tinkerer ESP 8266-12F
  MAX485

IDE options:

WeMos D1 R1
  4M(1M SPIFFS)
  80Mhz

Lichtspieler, 2019-06-17:
Hello,
I´ve the same problem.

Barabba11, 2020-01-31:
Hi, me too, I've found the solution and works, I used another project based on this one. Soon I'll tell how

Cysign, 2020-02-13:
@Barabba11: Why don't ya just tell now? I've finished my DMX PCB3 month ago....and couldn't manage to get it up and running in my network...

Barabba11, 2020-02-13:
Sorry I forgot to write here, I posted already the solution in next issue: https://github.com/mtongnz/ESP8266_ArtNetNode_v2/issues/124#issuecomment-584823019

Cysign, 2020-02-13:
Thanks ;)
I'll have a look at my setup ;)