mtongnz/ESP8266_ArtNetNode_v2 #105: getting unstable when setting both A/B ports to WS2812

PanOdbytek, 2018-09-13:
Hey! 
First of: big thanks for your work! It's amazing!

So i am using a NodeMcu v3 to run the ESP8266_ArtNetNeode_v2. When i choose the WS2812 mode in both A/B ports (and connect the led strips respectively to the TX and D4 pins), the board seems to go crazy, constantly loosing connection or restarting... 

Both ports are set to output data for 680 pixels. When i set up only one port, everything works fine.

Is that a known issue? Or maybe i am missing something?

thanks!

eplatero, 2018-10-17:
I'm having the same issue, single channel works great, but when I put strips on the secondary port, the board seems to hate it.

ozett, 2019-02-18:
powersupply too _small_ ?