mtongnz/ESP8266_ArtNetNode_v2 #109: BPM detection

Cysign, 2018-11-05:
As far as I kno that ESP2866 has an analog in.
As some Arduino projects for sound detection are based on the MSGEQ7 I wonder if a bpm-detection for stand-alone-mode would be possible.

If ArtNetNode might be able to save a programm adding a microphone and a MSGEQ7 to the ESP2866 might be the key for BPM-detection - which is the only thing I might miss on the ArtNetNode.

How difficult might the software-extension be?

ScholliYT, 2018-12-22:
This would be more of anohter project if I get your idea right. What you want is a DMX outputting device that has some BPM detection with microphone, right?

I did something similar with sofware sided bpm detection. So I'm just sending over the DMX data (what this project is for) and do the sound2light on my pc.

Cysign, 2018-12-26:
It's more like an externsion. I'd like to have stored scenes/chases which respond to BPM.
I wouldn't build it as an indepndent or a softwarebased system.

ulri-me, 2022-11-18:
The main problem you have is that bpm detection requires a constant polling of the ADC value coming in to check for the amplitude peaks in sound intensity.

This is not possible as the target hardware has a single core cpu only which stops the actual code execution constantly to perform the assember code and output actual byte data.

The additional question I have as well is this:
What would you do with this value?
You would also need to perform a full rewrite of the whole stand-alone output library, as you would have to calculate the correct output step based on elapsed time since program startup and the current bpm measured.

Cysign, 2023-01-08:
In my band room there is everything connected using DMX. So setting an animated colorsheme while jamming or listening to music, it could be synchronized to the ambient sound ;)

So on an dualcore ESP32 this might be possible, as it has two cores. Maybe I'll get into it one day...