mtongnz/ESP8266_ArtNetNode_v2 #112: A way to use this project with ethernet

mrv96, 2019-01-14:
Hi, in order to make the connection more stable, it could be added an option that enable the use of an ethernet interface. If wifi is not strictly necessary is better to use cables usually, and this project is very fine.

LoneWalkerWolf, 2021-03-12:
Ethernet is very difficult to do. 
I have tried many times and failed. Maybe take a look at:
https://www.instructables.com/Arduino-Artnet-Node/

mrv96, 2021-03-13:
Maybe Jonas Arnold succeeded: https://github.com/JonasArnold/EthernetDmxNode_esp8266

bangadrum, 2021-04-06:
Does the project compile for ESP32 boards? If so then other ESP projects (WLED, ESPHome) have got ethernet working reliably on the ESP32 boards, which I'm led to understand is much simpler than ethernet for ESP8266 boards.

LoneWalkerWolf, 2022-05-07:
Just thought I would reopen this thread. We have developed an ethernet version of this project. You can purchase our version from our Tindie store. https://www.tindie.com/products/lonewalker/ethernet-artnet-dmx-node/

LoneWalkerWolf, 2022-05-07:
> Just thought I would reopen this thread. We have developed an ethernet version of this project. You can purchase our version from our Tindie store. https://www.tindie.com/products/lonewalker/ethernet-artnet-dmx-node/

We are expecting to release the source code early next year.

mrv96, 2023-05-03:
I found the repo (https://github.com/expanseElectronics/ethernetNodes), but source code is not included.
The same is for https://github.com/expanseElectronics/artnetnodes, where I found .bin but not the source code.

LoneWalkerWolf, 2023-05-05:
We've decided not to make the source code available online for our wireless nodes; you can, however, flash the `.bin` file to an ESP8266 and use our pinout doc located [here ](https://github.com/expanseElectronics/artnetnodes/blob/gen3/pinout.md)to wire up the MAX485. With our ethernet nodes, we can't release any binaries, as the MAC address of the ethernet IC is hardcoded into the code and would not work when flashing a different IC. 

Please send me an email at [alexander@expanseelectronics.com](mailto:alexander@expanseelectronics.com), and we can send you the source code for our ethernet or wireless nodes.