mtongnz/ESP8266_ArtNetNode_v2 #113: Ajax "ambiguous overload for 'operator~' (operand type is 'IPAddress')"

Hugo500A, 2019-03-28:
I get this error when trying to compile espArtnetNode_2.0.0_b5g.ino

What am I doing wrong?

In file included from C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:41:0:

C:\Users\Hugo\Documents\Arduino\libraries\espArtnetRDM/espArtNetRDM.h: In member function 'void esp8266ArtNetRDM::init(IPAddress, IPAddress, bool, uint16_t, uint16_t, uint8_t*)':

C:\Users\Hugo\Documents\Arduino\libraries\espArtnetRDM/espArtNetRDM.h:157:75: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       init(ip, sub, dhcp, "espArtNetNode", "espArtNetNode", oem, esta, mac);

                                                                           ^

C:\Users\Hugo\Documents\Arduino\libraries\espArtnetRDM/espArtNetRDM.h:157:75: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Hugo\Documents\Arduino\libraries\espArtnetRDM/espArtNetRDM.h: In member function 'void esp8266ArtNetRDM::init(uint16_t, uint16_t, uint8_t*)':

C:\Users\Hugo\Documents\Arduino\libraries\espArtnetRDM/espArtNetRDM.h:168:93: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       init(INADDR_NONE, INADDR_NONE, false, "espArtNetNode", "espArtNetNode", oem, esta, mac);

                                                                                             ^

C:\Users\Hugo\Documents\Arduino\libraries\espArtnetRDM/espArtNetRDM.h:168:93: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino: In function 'void ipHandle()':

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:480:59: warning: narrowing conversion of '((~(int)deviceSettings.StoreStruct::subnet.IPAddress::operator[](0)) | ((int)(deviceSettings.StoreStruct::ip.IPAddress::operator[](0) & deviceSettings.StoreStruct::subnet.IPAddress::operator[](0))))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

     deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                           ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:480:138: warning: narrowing conversion of '((~(int)deviceSettings.StoreStruct::subnet.IPAddress::operator[](1)) | ((int)(deviceSettings.StoreStruct::ip.IPAddress::operator[](1) & deviceSettings.StoreStruct::subnet.IPAddress::operator[](1))))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

     deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                                                                                                          ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:480:217: warning: narrowing conversion of '((~(int)deviceSettings.StoreStruct::subnet.IPAddress::operator[](2)) | ((int)(deviceSettings.StoreStruct::ip.IPAddress::operator[](2) & deviceSettings.StoreStruct::subnet.IPAddress::operator[](2))))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

     deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                                                                                                                                                                                         ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:480:296: warning: narrowing conversion of '((~(int)deviceSettings.StoreStruct::subnet.IPAddress::operator[](3)) | ((int)(deviceSettings.StoreStruct::ip.IPAddress::operator[](3) & deviceSettings.StoreStruct::subnet.IPAddress::operator[](3))))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

     deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                                                                                                                                                                                                                                                                        ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\ajax.ino: In function 'bool ajaxSave(uint8_t, ArduinoJson::JsonObject&)':

ajax:101:55: error: ambiguous overload for 'operator~' (operand type is 'IPAddress')

       deviceSettings.broadcast = deviceSettings.ip | (~deviceSettings.subnet);

                                                       ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\ajax.ino:101:55: note: candidates are:

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\ajax.ino:101:55: note: operator~(uint32_t {aka unsigned int}) <built-in>

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\ajax.ino:101:55: note: operator~(u32_t {aka long unsigned int}) <built-in>

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\ajax.ino:101:55: note: operator~(int) <built-in>

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\startFunctions.ino: In function 'void artStart()':

[... 51 lines cut ...]

       deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                             ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\startFunctions.ino:383:140: warning: narrowing conversion of '((~(int)deviceSettings.StoreStruct::subnet.IPAddress::operator[](1)) | ((int)(deviceSettings.StoreStruct::ip.IPAddress::operator[](1) & deviceSettings.StoreStruct::subnet.IPAddress::operator[](1))))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

       deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                                                                                                            ^

C:\Users\Hugo\Documents\Arduino\espArtnetNode_2.0.0_b5g\startFunctions.ino:383:219: warning: narrowing conversion of '((~(int)deviceSettings.StoreStruct::subnet.IPAddress::operator[](2)) | ((int)(deviceSettings.StoreStruct::ip.IPAddress::operator[](2) & deviceSettings.StoreStruct::subnet.IPAddress::operator[](2))))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

       deviceSettings.broadcast = {~deviceSettings.subnet[0] | (deviceSettings.ip[0] & deviceSettings.subnet[0]), ~deviceSettings.subnet[1] | (deviceSettings.ip[1] & deviceSettings.subnet[1]), ~deviceSettings.subnet[2] | (deviceSettings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

[... cut ...] ings.ip[2] & deviceSettings.subnet[2]), ~deviceSettings.subnet[3] | (deviceSettings.ip[3] & deviceSettings.subnet[3])};

                                                                                                                                                                                                                                                                                                          ^

exit status 1
ambiguous overload for 'operator~' (operand type is 'IPAddress')

nikanet, 2019-04-10:
The same error

nikanet, 2019-04-10:
I change in ajax.ino
line 101:
deviceSettings.broadcast = deviceSettings.ip | (~deviceSettings.subnet);

to:
deviceSettings.broadcast = uint32_t(deviceSettings.ip) | (~uint32_t(deviceSettings.subnet));

and compiling now Done. Tested on Wemos D1 mini.

Lutzion, 2020-04-16:
The error depends on the used expressif version. You have to use a version < 2.x.
You also have to use an ArduinoJson-version < 6.x.

As i work with VSCode and platformio the resulting platformio.ini is:

```
[env:d1_mini]
platform = espressif8266@1.8.0   ; Version 2.4 not working, use the last 1.x version
board = d1_mini
board_build.f_cpu = 160000000L
;board_build.ld = eagle.flash.4m3m.ld
framework = arduino
build_flags = -ggdb3 ;-D ONE_PORT
lib_deps = 
    64@5.13.4  ; ArduinoJson -> Version 5 needed, so use last 5.x version
```

LoneWalkerWolf, 2021-03-12:
Yes, @Lutzion is correct. 
To quickly get rid of the errors. Use the following library versions:
- ESP8266 Version 2.4.1
- ArduinoJson Version 5.13.*

This works really well for a quick fix.