mtongnz/ESP8266_ArtNetNode_v2 #119: NodeMCU resets wifisettings after power cycle

Cysign, 2019-09-22:
Hey fellows,
I'm trying to get an NodeMCU running for this project.

Every time I power-cycle the device, it forgets the wifi settings and won't be able to connect to my wifi anymore :(

In the guy there's the green saves-button after I changed my wifi-settings.

What might be wrong with it?
Might the eeprom be broken?

Hitower78, 2020-04-05:
After get this devices Running, thats also my problem :-(

after save perhaps a reboot?

djkad, 2022-01-23:
Hi, I'm a developer. Although i'm new with nodemcu language, since i need a stable and minimal version of this project i'll put the hands inside this code. 
I needn't the webserver and other stuff. I will hardcode the settings into variables so the logic will be:
Loop forever looking the xxxx ssid.
When found just connect using yyyy password and start to receive/send artnet traffic using hardcoded settings like ip,name,ssid,password and modes. I think that having a minimal solution could be the only road to find serious bugs

djkad, 2022-01-23:
I succesfully made a working project. I have only to debug and solve a reset situation (probably because the watchdog timer). 
It's very fast because there is no webserver and it directly take the parameters at boot time without loading from filesystem. I was having problems with settings too but finally no more clear settings problems. By the way it's very easy to understand and it uses only 2 libraries: Wifi and Fastled. I directly read udp packet without having artnet libraries. When (or if) i find the solution of the reset i will share the code with mtongnz and you.

Cysign, 2022-01-23:
Sounds good! I still have the hardware next to my desk...but I gave up trying years ago :D