mtongnz/ESP8266_ArtNetNode_v2 #123: DMX in for Fx

kefkevm, 2019-12-19:
Hi, 

I want to use a DMX signal from a external controler to control ws2812 leds with the FX mode of the ESP Node.  How can i do this?

emilab, 2020-01-14:
Hello!!you already tried?

kefkevm, 2020-01-14:
Yes i tried

emilab, 2020-01-15:
🙄and??

emilab, 2020-01-16:
Hello @kefkevm

kefkevm, 2020-01-16:
It Does not work. Set the A connection to receive DMX. And B to ws2812 with 12 FX. But my leds wil not light up..

funkmasterfrank, 2021-11-22:
I have everything working except randomly it seems to lose connect to the other esp8266 with Wled installed. The first time it worked for 1 hour so I thought it was overheating in the case so I removed the cover and it worked for about 5 hours then it disconnected or something. Is there a sleep mode causing this issue? Any help would be appreciated

tapi35, 2022-12-04:
Hello, Funkmasterfrank, I want to create something liké you with two wemos. 
First, stairville led foot 8 DMX connected on wemos with artnetnode / wireless transmission / wemos with wled / led strip.
Have you resolved this problem ? Do you use it ?  Is everything works fine  ? 
Thanks

emilab, 2022-12-05:
Yes it's work fine!!

Ottenere Outlook per Android<https://aka.ms/AAb9ysg>

tapi35, 2022-12-05:
I stopped working on my project since many years so I start like a newbie. I have flashed two wemos, One configurated to receive artnet with dmx out. It works great with My laptop and Qlc. The Other with Port A =>dmx in. Exactly the same connection ( TX GPIO). The XLR is in the output of my dmx pedal hardware. Normaly in qlc (plugin Artnet / Entry 192.168.1.127), I should see faders moving when I change my hardware presets ?
I don't understand what is wrong !!!! 

[image]
[image]

If I've made a mistake, thanks a lot for your help !

tapi35, 2022-12-07:
If someone could help me, it will be great  ! I have spent 10 hours on it and I don't know if it's problem of connection or configuration. 
My wled works fine with qlc on artnet but I don't know if my ArtnetNode works !
I post screenshots and schema.....Thanks a lot

[image]
[image]
[image]
[image]
[image]
[image]

kefkevm, 2022-12-08:
I think your broadcast address should be the ip of your router or pc where Qlc is running on. I am not sure about this, you have to try it...