mtongnz/ESP8266_ArtNetNode_v2 #124: Any Project progress / new more stable FW version

Nostromo-star, 2020-01-04:
Hi,

this is still an interesting project and I wonder is there is still activity
respectively work on a more stable ESP8266 firmware that is addressing
the known bug in the code.

thanks guys,
TH

mrv96, 2020-02-03:
In my opinion the only relevant progress was done by JonasArnold in his fork, but he never make a pull request. I tried his firmware and for me its stability is better respect the original one.

I "forked" his repo and I fixed the compilation errors related to the latest ESP8266 core version. You can find my repo here: https://github.com/mrv96/ESP8266_ArtNetNode_v2/tree/JonasArnold

Barabba11, 2020-02-11:
For me it worked this, and this only after I tried many others.
It's based on the same project, so no evident differences:
https://github.com/bombcheck/ESP8266_ArtNet-LED-DMX-Node

mrv96, 2020-02-11:
@Barabba11 why is the project that you linked better? Which are the differences with respect the original one?

Barabba11, 2020-02-11:
Simply, it works :) one week past and a bit hard to remember, what I can say:
the port A don't produce any output on port TX0 (the other yes), only port B outputs, but probably it has a jitter problem. I was never able to see any reaction on fixtures from port B.
The bombcheck I connected to port A and don't test the port B, so I can't say if the problem was only that the TX pin was not outputing

djkad, 2022-01-23:
Hi, I'm a developer. Although i'm new with nodemcu language, since i need a stable and minimal version of this project i'll put the hands inside this code.
I needn't the webserver and other stuff. I will hardcode the settings into variables so the logic will be:
Loop forever looking the xxxx ssid.
When found just connect using yyyy password and start to receive/send artnet traffic using hardcoded settings like ip,name,ssid,password and modes. I think that having a minimal solution could be the only road to find serious bugs