mtongnz/ESP8266_ArtNetNode_v2 #126: Problem for flashing code with d1 mini

tapi35, 2020-03-15:
Hello, 
I want to use d1 mini for this code for connecting dmx and ws2812. I don't succeed in using arduino IDE. I have flash "espArtnetNode_2.0.0_b5g_WEMOS.bin"  with ESP.Easy.Flasher.  I have the web interface on hotspot mode, I configure my ssid with password but when I turn off the d1, all the parameters are cancelled. That's the first problem. 
Is it possible to use arduino IDE directly with the d1 mini connected via usb ? 
I have test with a lot of configuration : put LOLIN(WEMOS) d1 r2 and mini, flash size 4M(3M SPIFFS), cpu in 160Mhz. I have test with blink, it's ok but with the espArtnetNode2.0.0-b5g, message wifistart was not declare on this scope". 
I don't use level shifter for the moment, is it a problem ?
Sorry for all those questions but I' m totally lost.  Thanks a lot for your help

muxlux, 2020-03-15:
I had a similar experience with D1 Mini until flashing an earlier version https://github.com/mtongnz/ESP8266_ArtNetNode_DMX.git
I was then able to upgrade/update to ESP8266_ArtNetNode_v2 through Web UI OTA

tapi35, 2020-03-16:
Thanks a lot ! it works. It's write on the description but I didn't notice it.

muxlux, 2020-03-16:
glad it's working!
i connected mine to a $2 max485 module & it's been working well with 90
channels (15 fixtures)
have fun!

Hitower78, 2020-03-31:
Hi ther!
i want to try it on a d1 mini, but i found a couple of Problems with the upload.
The V2 and V1 version doesnt load up.
i get errors in die IDE (1.8.12): StoreInit was not declared...
and several Artnet errors "sendartnetreplay" was not declared...

Did i miss a file or something? i am New to that stuff buw other projects are working fine :-)

muxlux, 2020-03-31:
are you compiling it for the 'LOLIN(WEMOS) D1 R2 & mini' board?

muxlux, 2020-04-02:
i'm using Arduino 1.8.10
may want to try downgrading

Hitower78, 2020-04-05:
Hallo zusammen
Hi there,

i use to flash the BIN, thats very easy with no complications!

in IDE the errors are still there... but the BIN is running now!

thanks

Hoizna, 2020-05-03:
is the dmx output of your nodes stucking (not smooth) ? mine is stucking

Hitower78, 2020-05-04:
i never check the DMX Out. 
That would by the case in my next Project

muxlux, 2020-05-04:
i've experienced occasional signal interruptions
if it's real bad, you may have a & b on the rs485 module/transceiver

muxlux, 2020-05-04:
...reversed

Hoizna, 2020-05-04:
I have everything built like in the schematic... maybe it is because i am working with expensive nodes at work. 
But i discovered, that the Old version: 1.2.0 is much more stable, and i don´t have problems with it. I don´t need the dmx in and the scene storage is also a good feature to have, so i will probably switch back again to the old version... Maybe the v2 gets more stable in some future code releases, than i will try again. Thanks for your help!

muxlux, 2020-05-04:
i had better stability with the older version too
i was only using one dmx output

it's a great project but there're quite a few open issues & the last commit
was two years ago
might be worth checking out some of the forks

Daniel091988, 2021-09-15:
Hello 

I Flashed the Wemos D1 mini and connected it according one of the schematics.
but I dont get an DMX output. I am not sure if everything is connected correctly because the schematics is not for the wemos and i tried to connect it according the table. but its not working. 
Can someone tell me how to connect the Wemos D1 mini correct with the Max 485 Module?

Thank you

muxlux, 2021-09-16:
Have you swapped tx & rx? I've made that mistake before.

Daniel091988, 2021-09-16:
I found the problem today. The dmx device did not recognize the dmx. 
i used DMX Monitor app to send a signal via artnet and now it shows a signal.