mtongnz/ESP8266_ArtNetNode_v2 #127: I'm almost there but....

tapi35, 2020-03-25:
Hello, Thanks for tjis great project ! I'm musician and I work on it for my show. It works for me with dmx hardware controlled by QLC+ with d1 mini. Now I build a 75X8 matrix of ws2812 using jinx. There's two days that I have a problem to solve ( image below)
I want to create 4 universes : 0,1,2,3 in jinx, each universe controls two lines of 75 pixels (so 150 per universe or 450 dmx channels). but in the panel, there is a hole of 20 pixel on the universe 1 and 2, the first : number 0 is ok
I don't understand. I 've tested with port a et port b, it's the same. I have also tested my strip directly and It seems to be ok. I have tested with many outpatch different, I have changed the number of pixel by universe in the port A settings.....the same.

is that somebody has already had this problem ?
Thank you
[image]

tapi35, 2020-03-27:
In fact, I realize there's only universe 0 which works. I have tested with just one line of leds and in jinx 3 pixel per universe on the same line. There's just the 3 pixels of the universe 0. 
There is something that I don't understand !!!  Is it ok that 4 universes are on the same connection (tx for port A or d4 for port B  on the d1 mini ) ?

sparkescreative, 2021-05-09:
I seem to have the same problem. I have a gap of 20 pixels/LEDs between universes. Did you get it sorted?