mtongnz/ESP8266_ArtNetNode_v2 #128: only one universe is working

tapi35, 2020-03-30:
I have made a new test with just one strip of 75 pixel with my d1 wemos
It works great with dmx on my devices but with strip led, only the 0 universe (0.1.2.3) works. 
If I connect only one cable between the D4 pin and the the strip, I can address 4 universes on the same strip ? the d4 pin, alone, is able to drive the 4 universes, we are ok with that ?
I'm becoming crazy.....
Or do you think it could be a problem of flash? I have used ESP.Easy.Flasher, put "espArtnetNode_2.0.0_b5g_WEMOS" and it works (wifi, hotspot, dmx).
Is it possible that only the led library is missing ? When you flash directly, libraries are included ? I can't choose 160 MHz with this method so I don't know if it could change anything ?
I'm sorry for asking a lot of questions on this forum but I don't know anyone who could help me.  Thanks a lot  to those who can help me.

tapi35, 2020-04-04:
Nobody could help me ? or wants to help me ? Is it because my english is very bad or because I'm too stupid ( I already know that ;-) or I'm the first with this problem ?
I continue to test with differents settings in jinx and the hotspot mode. Perhaps, I will be lucky one day ....until it works.  it's just so frustating  rrrrrr

tapi35, 2020-04-09:
I try directly in qlc and only the universe 0 on TX works and the universe 4 on d4. Before, I've flashed a blank firmware to reset and reinstall a new one. Always the same problem.

OneDayFlie, 2020-04-18:
Same problem here... working with a 16x16 panel... and if i patch the second universe on it... i will miss a few rows and everything in the second universe is scrambled...

LoneWalkerWolf, 2021-03-12:
This is quite simply an issue with the ESP8266 chip itself. 
It simply does not have enough RAM to store all the incoming ArtNet data it is receiving. So it just rewrites over the incoming data. 
If you really want to control more than one universe at a time with WS2812 mode, it would be much easier to buy a cheap destripilizer from AliExpress and just use the node in DMX mode to connect to it. 
Limited functionality, I know, but it's just a workaround until we can use the ESP32. I'm working on this now :)