mtongnz/ESP8266_ArtNetNode_v2 #129: set correct flash properties in arduino IDE

mschlen, 2020-03-30:
Hello community,,

first thanks for this great project, I really love it. I only wanted to change a few things by myself in the code (like Wifi-Settings or add a few lines of code to my project needs).

I have a problem with the flashing of a ESP 8266 07S. The arduino IDE is working and compiling without a problem. Also flashing to the ESP does not throw an error. But the problem is, that the ESP does not open as a Wifi-device. If I am flashing the bin-file with the NODEMCU Firmware Programmer, it works without a problem Due to that, I can prove the hardware is working as expected. If using the compiled Arduino .bin file and try to flash it with the NODEMCU Firmware Programmer, the same issue appears - flashing is complete, but no Wifi Hotspot to log in.

So I think, there must be something wrong with my settings in the Arduino Tools. Please see attached file for my settings.

[image]

For me it is equal, if i compile in Arduino IDE and upload the bin. file with the NODEMCU Programmer, or directly compile and upload with the IDE. The essential would be, that it works.

I hope, anyone can tell me, if there is something wrong.

Thank you all,
Markus

mschlen, 2020-06-11:
So no comments?

LoneWalkerWolf, 2021-03-12:
Try change the following:
- CPU Frequency = 160Mhz
- Crystal Frequency = 26Mhz
- Flash Mode = DOUT
- Reset method = nodemcu

If none of that works, try changing the ESP8266 board library to version: 2.4.1
This also helps with compiling errors. 
If you are still having problems, I am always happy to just send you a .bin file for you to flash manually.