mtongnz/ESP8266_ArtNetNode_v2 #141: Turn LEDs off on startup

MasterSmollett, 2022-06-13:
Hi! Fantastic work!

Is it possible to turn all led's (ws2812b) **off** on startup ? Or not turn them on ?

At the moment all led's turn white, I would like to avoid the power surge.

My cpp is a bit rusty, but I'm happy to do a PR if you point me in the right direction.

Thanks!

ulri-me, 2022-11-18:
Hi,

I have never worked with this lib and the WS2812b LEDs.

But have a look at this reddit discussion:
https://www.reddit.com/r/arduino/comments/d5yjmt/ws2812b_default_color_after_power_on/

The question is if the library does anything itself to start the LEDs with full white.

As far as I can see the only DMX output buffer initialisation with a default value is this line
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/blob/4a1f803e6d3cd0946d18dc851ab2eeb200e828f4/libs/espDMX_RDM/espDMX_RDM.cpp#L384
which stores 0 as default value for a newly created DMX buffer.

And for the LED strip itself does this line
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/blob/4a1f803e6d3cd0946d18dc851ab2eeb200e828f4/source/startFunctions.ino#L113
execute this function
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/blob/4a1f803e6d3cd0946d18dc851ab2eeb200e828f4/source/ws2812Driver.cpp#L29
which also clears the buffer to 0's by calling this function
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/blob/4a1f803e6d3cd0946d18dc851ab2eeb200e828f4/source/ws2812Driver.cpp#L70

ulri-me, 2022-11-18:
So I would assume you have to be faster with starting the output (or even start the output before powering up the LED strip) or get this capacitor trick from Reddit to work.