mtongnz/ESP8266_ArtNetNode_v2 #143: Random Crashes

gmint, 2022-11-28:
I have two ESP8266 Node MCU devices. One of them is driving about 450 pixels out of one output and about 120 out of the second output. The second device is driving about 300 pixels out of one output. I am experiencing random hangs on both devices. Sometimes I can go hours without anything hanging, other times it will be 20 minutes. I've tried everything I can think of: e.g. increasing the size of the capacitors, making sure I have a very large, stable, and correctly calibrated power supply, making sure my LED strips have been power injected at the required points, making sure my wifi access point is located very close to the ESPs.

These are both being driven by ArtNet. I'd considered trying sACN, but have not done this yet. Are there any other tips or tricks that I might try to increase the stability?

madis25, 2023-03-20:
The same there. Was thinking that it may be router problem, but the same is with other routers. 4 ESP-s on table, turned on same time, working, then one by one falling down. Just power off works sometimes, most of the times getting hotspot mode. Really annoying to fill again all data :(

madis25, 2023-11-04:
But I remember that I made this priject first time in january 2019. Then I hadnt this kind of problems. Maybe libraries updated/changed or something?

517002650, 2023-12-27:
I have also encountered this problem. Controlling WS2812 will cause this problem, requiring two restarts to control the output of the DMA. It is normal

517002650, 2024-01-09:
> 但我记得我在 2019 年 1 月第一次做了这个 primeject。然后我就没有这种问题了。也许库更新/更改了什么？

Translation: Do you remember the library version at that time and do you have any impression?